FOSSEE/eSim #45: eSim splash screen hides the workspace dialoge box on launch

kavyamanohar, 2015-08-25:
On Ubuntu 15.04 operating system upon launching eSim, the splash screen hides the dialogue box which prompts the user to choose the workspace location. Unless the workspace location is not chosen eSim wont be launched. User unaware of the dialogue box in the background might have to wait endlessly for eSim to launch. However upon a mouse click the splash screen  quits itself and  the workspace dialogue box becomes visible . 

The splash screen should either disappear after a time out period or the the workspace dialogue box should be above the splash screen.

fahim-oscad, 2015-08-25:
Thanks for reporting issue and giving suggestion. I have tested it on Ubuntu 12.04,14.04 and Windows 7. And in my machine the workspace window is coming on the top of splash screen. 

Let me check if the issue is only with 15.04.

fahim-oscad, 2015-08-28:
I check it on Ubuntu 15.04 and its working perfectly. Looks like its machine specific.

kavyamanohar, 2015-09-02:
Sorry, I missed to mention that the above issue was while launching eSim with KDE on Ubuntu 15.04.
It works well with Unity.

fahim-oscad, 2015-09-07:
Thank you for your valuable inputs. We will definitely change it in next version.

closed 2015-09-07 by fahim-oscad