FOSSEE/eSim #58: Libraries not found

cryarchy, 2016-06-17:
I have installed eSim on my computer (Linux Mint 17.3 Cinnamon 64-bit) and the installation was successful. However, creating a new project and opening the schematic causes a dialog with the title `Not Found` and the following content to pop-up:
```
The following libraries were not found:
- adc
- dac
- memory
- xilinx
- microcontrollers
- dsp
- microchip
- analog_switches
- motorola
- texas
- intel
- audio
- interface
- digital-audio
- philips
- display
- cypress
- siliconi
- opto
- atmel
- contrib
- power
- device
- transistors
- conn
- linear
- regul
- 74xx
- cmos4000
- eSim_Analog
- eSim_Devices
- eSim_Digital
- eSim_Hybrid
- eSim_Miscellaneous
- eSim_Power
- eSim_Sources
- eSim_Subckt
- eSim_User
- eSim_Plot
- eSim_PSpice
```

The `choose component` dialog box also indicated that **0 components loaded** and no components are available. Any help on how to resolve this will be greatly appreciated.

cryarchy, 2016-06-18:
I partially fixed the issue by downloading and adding the KiCAD libraries from https://github.com/KiCad/kicad-library. This now resulted in the error,
```
The following libraries were not found:
- eSim_Analog
- eSim_Devices
- eSim_Digital
- eSim_Hybrid
- eSim_Miscellaneous
- eSim_Power
- eSim_Sources
- eSim_Subckt
- eSim_User
- eSim_Plot
- eSim_PSpice
```
I added the libraries that are contained in the `eSim\kicadSchematicLibrary\` and now I have the error(s):

```
08:36:44 AM: Library 'eSim_PSpice' has duplicate entry name 'R_var_PSPICE'.
This may cause some unexpected behavior when loading components into a schematic.
08:36:44 AM: Library 'eSim_PSpice' has duplicate entry name 'R_var_PSPICE'.
This may cause some unexpected behavior when loading components into a schematic.
08:36:44 AM: Library 'eSim_PSpice' has duplicate entry name 'C_var_PSPICE'.
This may cause some unexpected behavior when loading components into a schematic.
08:36:44 AM: Library 'eSim_PSpice' has duplicate entry name 'C_var_PSPICE'.
This may cause some unexpected behavior when loading components into a schematic.
08:36:44 AM: Library 'eSim_PSpice' has duplicate entry name 'DB9M_PSPICE'.
This may cause some unexpected behavior when loading components into a schematic.
08:36:44 AM: Library 'eSim_PSpice' has duplicate entry name 'DB9M_PSPICE'.
This may cause some unexpected behavior when loading components into a schematic.
08:36:44 AM: Library 'eSim_PSpice' component load error Wrong DEF format in line 3184, skipped..
08:36:44 AM: Library 'eSim_PSpice' component load error Wrong DEF format in line 10037, skipped..
08:36:44 AM: Library 'eSim_PSpice' component load error Wrong DEF format in line 10052, skipped..
08:36:44 AM: Library 'eSim_PSpice' component load error Wrong DEF format in line 10067, skipped..
08:36:44 AM: Library 'eSim_PSpice' component load error Wrong DEF format in line 10082, skipped..
08:36:44 AM: Library 'eSim_PSpice' component load error Wrong DEF format in line 10097, skipped..
08:36:44 AM: Library 'eSim_PSpice' component load error Wrong DEF format in line 10112, skipped..
08:36:44 AM: Library 'eSim_PSpice' component load error Wrong DEF format in line 10127, skipped..
08:36:44 AM: Library 'eSim_PSpice' component load error Wrong DEF format in line 10142, skipped..
08:36:44 AM: Library 'eSim_PSpice' component load error Wrong DEF format in line 10157, skipped..
08:36:44 AM: Library 'eSim_PSpice' component load error Wrong DEF format in line 10172, skipped..
08:36:44 AM: Library 'eSim_PSpice' component load error Wrong DEF format in line 10187, skipped..
08:36:44 AM: Library 'eSim_PSpice' component load error Wrong DEF format in line 10205, skipped..
08:36:44 AM: Library 'eSim_PSpice' component load error Wrong DEF format in line 10220, skipped..
08:36:44 AM: Library 'eSim_PSpice' component load error Wrong DEF format in line 10235, skipped..
08:36:44 AM: Library 'eSim_PSpice' component load error Wrong DEF format in line 10248, skipped..
08:36:44 AM: Library 'eSim_PSpice' component load error Wrong DEF format in line 10261, skipped..
08:36:44 AM: Library 'eSim_PSpice' component load error Wrong DEF format in line 10273, skipped..
08:36:44 AM: Library 'eSim_PSpice' component load error Wrong DEF format in line 10285, skipped..
08:36:44 AM: Library 'eSim_PSpice' component load error Wrong DEF format in line 10297, skipped..
08:36:44 AM: Library 'eSim_PSpice' component load error Wrong DEF format in line 10308, skipped..
08:36:44 AM: Library 'eSim_PSpice' component load error Wrong DEF format in line 10319, skipped..
08:36:44 AM: Library 'eSim_PSpice' component load error Wrong DEF format in line 10337, skipped..
08:36:44 AM: Library 'eSim_PSpice' component load error Wrong DEF format in line 10355, skipped..
08:36:44 AM: Library 'eSim_PSpice' component load error Wrong DEF format in line 10371, skipped..
08:36:44 AM: Library 'eSim_PSpice' component load error Wrong DEF format in line 10386, skipped..
08:36:44 AM: Library 'eSim_PSpice' component load error Wrong DEF format in line 10401, skipped..
08:36:44 AM: Library 'eSim_PSpice' component load error Wrong DEF format in line 10413, skipped..
08:36:44 AM: Library 'eSim_PSpice' component load error Wrong DEF format in line 10428, skipped..
08:36:44 AM: Library 'eSim_PSpice' component load error Wrong DEF format in line 10441, skipped..
08:36:44 AM: Library 'eSim_PSpice' component load error Wrong DEF format in line 12053, skipped..
08:36:44 AM: Library 'eSim_PSpice' component load error Wrong DEF format in line 12133, skipped..
08:36:44 AM: Library 'eSim_PSpice' component load error Wrong DEF format in line 12155, skipped..
08:36:44 AM: Library 'eSim_PSpice' component load error Wrong DEF format in line 12180, skipped..
08:36:44 AM: Library 'eSim_PSpice' component load error Wrong DEF format in line 12261, skipped..
08:36:44 AM: Library 'eSim_PSpice' component load error Wrong DEF format in line 12273, skipped..
08:36:44 AM: Library 'eSim_PSpice' component load error Wrong DEF format in line 12286, skipped..
08:36:44 AM: Library 'eSim_PSpice' component load error Wrong DEF format in line 12299, skipped..
08:36:44 AM: Library 'eSim_PSpice' component load error Wrong DEF format in line 12431, skipped..
08:36:44 AM: Library 'eSim_PSpice' component load error Wrong DEF format in line 12444, skipped..
```
How do I fix this?

kavyamanohar, 2016-06-18:
The  libraries can be added manually. To add libraries Go to `Preferences -> component libraries -> add` . Choose the required libraries from `user/share/kicad/library`.
The libraries other than the `eSim_*` comes from kicad. If they are not found in the above location, Installing `kicad-library` package might be required.

cryarchy, 2016-06-18:
Thanks Kavya, I've fixed the issue with the libraries by installing and adding the libraries from `kicad-library`. Is there a current fix to the resulting `eSim_PSpice` errors detailed above?

fahim-oscad, 2016-06-18:
I can't find any duplicate element in eSim_PSpice. Can you check kicad.pro file and see if eSim_PSpice has been used twice ? It can be found under /usr/share/kicad/template.

cryarchy, 2016-06-18:
No it hasn't. There is only one entry for eSim_PSpice. Here is the file. I've added the `.txt` extension so that it may upload.
[kicad.pro.txt](https://github.com/FOSSEE/eSim/files/321629/kicad.pro.txt)

fahim-oscad, 2016-06-22:
In Ubuntu if we run the eSim installer it install kicad properly and create the library folder as well however in Linux Mint 17.3 (Cinnamon 64-bit) we found that library directory was not creating while installing Kicad and it has to separately install. Our installer just copy few files in library and template folder.

Here is the solution for Linux Mint after installing eSim just install kicad-library.
`sudo apt-get install kicad-library`
 This will create kicad library folder

Now just copy the .lib and .dcm files available in kicadSchematicLibrary of eSim into library folder

`sudo cp -r kicadSchematicLibrary/*.lib /usr/share/kicad/library/`
`sudo cp -r kicadSchematicLibrary/*.dcm /usr/share/kicad/library/`

We have tested this procedure in Mint and it is working. In the next version we will add the kicad-library installation as well.

cryarchy, 2016-06-22:
Thank to all you guys!

closed 2016-06-23 by fahim-oscad

Mike6158, 2017-09-29:
I solved the problem by uninstalling KiCad and forgetting it ever existed