FOSSEE/eSim #59: Simulation window startup error

cryarchy, 2016-06-18:
Following the process detailed in the **eSim user manual** I am trying to simulate as detailed in chapter 6. The process is a success through to the kicadToNgSpice conversion. However, pressing on the *Simulation* button on the left-side tool bar on the eSim interface results in a window with the following info:
```
Error while opening python plotting Editor. Please look at console for more details 
```
The contents of the console are:
```
eSim Started......
Project Selected : None

[INFO]: Workspace : /home/user/tizy/eSim-Workspace
[INFO]: The current project is /home/user/tizy/eSim-Workspace/Project1
[INFO]: Kicad Schematic is called for project /home/user/tizy/eSim-Workspace/Project1
[INFO]: Kicad to Ngspice Conversion is called
[INFO]: Current Project is /home/user/tizy/eSim-Workspace/Project1
[INFO]: Ngspice simulation is called : /home/user/tizy/eSim-Workspace/Project1
[INFO]: PythonPlotting is called : /home/user/tizy/eSim-Workspace/Project1
[ERROR]: Exception Message :[Errno 2] No such file or directory: '/home/user/tizy/eSim-Workspace/Project1/plot_data_i.txt'
[ERROR]: Exception Message :local variable 'alli' referenced before assignment
[ERROR]: Exception Message : [Errno 2] No such file or directory: '/home/user/tizy/eSim-Workspace/Project1/plot_data_v.txt'
[INFO]: Ngspice simulation is called : /home/user/tizy/eSim-Workspace/Project1
[INFO]: PythonPlotting is called : /home/user/tizy/eSim-Workspace/Project1
[ERROR]: Exception Message :[Errno 2] No such file or directory: '/home/user/tizy/eSim-Workspace/Project1/plot_data_i.txt'
[ERROR]: Exception Message :local variable 'alli' referenced before assignment
[ERROR]: Exception Message : [Errno 2] No such file or directory: '/home/user/tizy/eSim-Workspace/Project1/plot_data_v.txt'
[INFO]: Ngspice simulation is called : /home/user/tizy/eSim-Workspace/Project1
[INFO]: PythonPlotting is called : /home/user/tizy/eSim-Workspace/Project1
[ERROR]: Exception Message :[Errno 2] No such file or directory: '/home/user/tizy/eSim-Workspace/Project1/plot_data_i.txt'
[ERROR]: Exception Message :local variable 'alli' referenced before assignment
[ERROR]: Exception Message : [Errno 2] No such file or directory: '/home/user/tizy/eSim-Workspace/Project1/plot_data_v.txt'
```
Any insight as to how I can resolve this?

cryarchy, 2016-06-18:
I was able to resolve this problem by first running `ngspice Project1.cir.out` while in the **root folder** of my project (named *Project1*). The simulation on *terminal ngspice* flops but it generates the missing files. I then returned to the eSim interface and clicking on the *Simulation* button on the left-side tool bar on the eSim interface worked out fine. I leave this issue opened since I believe that is not the intended workflow. It may be closed if that is the intended workflow.

fahim-oscad, 2016-06-18:
What version of eSim are you using ? Did you enter 'y' to install nghdl ?

cryarchy, 2016-06-18:
Yes I did. Version: `eSim-1.1.2`

fahim-oscad, 2016-06-20:
Please raise the example related query on our forum http://forums.fossee.in/   

Closing this issue.

closed 2016-06-20 by fahim-oscad