FOSSEE/eSim #82: Add feature to rename projects

nilshah98, 2019-06-10:
Add feature to rename projects, and also their respective files, in the selected folder.
Refer - 5e58022d8413f3ac6608cfac06ba91f3ee1a3840

sunilshetye, 2019-06-14:
Create three projects named **a**, **ab**, and **b**. Test cases for this issue:

1. Rename **a** to _blank_: Should fail saying cannot give empty project name.
2. Rename **a** to **a**: Should fail saying that project name has to be different.
3. Rename **a** to **b**: Should fail saying that **b** is an existing project.
4. Rename **a** to **c**: Only project **a** should be renamed to **c**. **ab** should not be renamed.

anjalijaiswal08, 2019-06-14:
Okay, will look into it. Thank you.

saurabhb17, 2019-06-20:
has this been resolved?

anjalijaiswal08, 2019-06-20:
> has this been resolved?

Yes, it has been solved, will push it today only. Thank you.

sunilshetye, 2019-06-27:
merged into fellowship2019-python3 branch

closed 2019-06-27 by sunilshetye