FOSSEE/eSim #159: xpdf depreciated on new Ubuntu versions

q-wertz, 2020-11-03:
When trying to install eSim on newer Ubuntu versions the installation of xpdf fails.
As stated [here](https://askubuntu.com/a/1245541) xpdf was dropped from Ubuntu with version 20.04.
Would it be possible for eSim to use another pdf reader?

saurabhb17, 2020-11-04:
yes of course, for Ubuntu 20.04 onward, we will use either evince or okular. Will be incorporated in next release

q-wertz, 2020-11-04:
Ok thank you very much.
I have quite some trouble installing it in Ubuntu 20.10. There are more dependencies which are broken.

saurabhb17, 2020-11-04:
You know, I have tried running eSim on Ubuntu 20 on a Virtualbox setup, surprisingly,I did not encounter the l_ibz.so.1: version `ZLIB_1.2.9_ issue.

Nevertheless, I will get back to you on this.

ThellraAK, 2020-11-15:
Manually installed xpdf from the 19.10 repos and then ran into not being able to  install gnat-5 and couldn't find a way to do the same (lubuntu 20.04)

rahulp13, 2021-01-08:
There is a new release that fixed all the above issues. It is available [here](https://esim.fossee.in/downloads). You can try it and let us know if there are any further issues.

q-wertz, 2021-01-08:
Thanks for the new release :)

The problem is, that you only support an old kicad version and this leads to the following problem(s)
```
The following additional packages will be installed:
  libboost-context1.65.1 libboost-system1.65.1 libboost-thread1.65.1 libglew2.0 libpython2-stdlib libpython2.7
  libpython2.7-minimal libpython2.7-stdlib libwxgtk3.0-0v5 python-is-python2 python-wxgtk3.0 python-wxversion
  python2 python2-minimal python2.7 python2.7-minimal
Suggested packages:
  extra-xdg-menus kicad-demo kicad-locale-ko | kicad-locale-ru | kicad-locale-pl | kicad-locale-pt | kicad-locale-ja
  | kicad-locale-id | kicad-locale-ca | kicad-locale-it | kicad-locale-el | kicad-locale-zh | kicad-locale-de
  | kicad-locale-sl | kicad-locale-cs | kicad-locale-bg | kicad-locale-sv | kicad-locale-lt | kicad-locale-fi
  | kicad-locale-fr | kicad-locale-hu | kicad-locale-nl | kicad-locale-es | kicad-locale-sk glew-utils wx3.0-doc
  python2-doc python-tk python2.7-doc
Recommended packages:
  kicad-library kicad-doc-en
The following packages will be REMOVED:
  kicad-footprints kicad-libraries kicad-symbols kicad-templates
The following NEW packages will be installed:
  libboost-context1.65.1 libboost-system1.65.1 libboost-thread1.65.1 libglew2.0 libpython2-stdlib libpython2.7
  libpython2.7-minimal libpython2.7-stdlib libwxgtk3.0-0v5 python-is-python2 python-wxgtk3.0 python-wxversion
  python2 python2-minimal python2.7 python2.7-minimal
The following packages will be DOWNGRADED:
  kicad
0 upgraded, 16 newly installed, 1 downgraded, 4 to remove and 0 not upgraded.
E: Packages were downgraded and -y was used without --allow-downgrades.

Error! Kindly resolve above error(s) and try again.

Aborting Installation...
```

Is it possible for you to use the new kicad version or are there API changes that you do not support?
I'm not a big fan of using python2 :sweat_smile:

rahulp13, 2021-01-08:
Some modules of eSim and libraries were developed for old KiCad version. So, I highly doubt its compatibility with the new KiCad version. You can manually downgrade it to proceed with the old version. Otherwise, you can proceed with the new KiCad version (comment out the KiCad installation part in the script) and let us know if it works or not. You may also send a pull request for the same, and we shall make it available in our next release.

Regarding Python2, Even I'm not a big fan of it, but it shouldn't do any harm to your usage of eSim.

closed 2021-01-31 by rahulp13