FOSSEE/eSim #164: Install script uses old version of kicad ppa

geraldsweers, 2021-07-06:
Script uses old version of kicad ppa
I have kicad5.1.9 installed already
So i changed it to the new one for myself

Made these changes:
---------------------------------------------------

function addKicadPPA
{

    #sudo add-apt-repository ppa:kicad/kicad-5.1-releases
    kicadppa="kicad/kicad-5.1-releases"
    findppa=$(grep -h -r "^deb.*$kicadppa*" /etc/apt/sources.list* > /dev/null 2>&1 || test $? = 1)
    if [ -z "$findppa" ]; then
        echo "Adding KiCad-5 PPA to local apt-repository"
        if [[ $(lsb_release -rs) == 20.* ]]; then
            sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 83FBAD2D910F124E
            sudo add-apt-repository --yes "deb [trusted=yes] http://ppa.launchpad.net/kicad/kicad-5.1-releases/ubuntu>
            sudo touch /etc/apt/preferences.d/preferences
            echo "Package: kicad" | sudo tee -a /etc/apt/preferences.d/preferences > /dev/null
            echo "Pin: version 4.0.7*" | sudo tee -a /etc/apt/preferences.d/preferences > /dev/null
            echo "Pin-Priority: 501" | sudo tee -a /etc/apt/preferences.d/preferences > /dev/null
            sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 3B4FE6ACC0B21F32
            sudo add-apt-repository --yes "deb http://archive.ubuntu.com/ubuntu/ bionic main universe"
        else
            sudo add-apt-repository --yes ppa:kicad/kicad-5.1-releases
        fi
    else
        echo "KiCad-5 is available in synaptic"
    fi

}

------------------------------------------------------------------
and i also changed:

   echo "Installing KiCad..........................."
    sudo apt-get install -y --no-install-recommends kicad=5.1.9*
    if [[ $(lsb_release -rs) == 20.* ]]; then
        sudo add-apt-repository -ry "deb http://archive.ubuntu.com/ubuntu/ bionic main universe"
    fi

---------------------------------------------------------------

Dont know how to send changes otherway. Im a newbe and afraid to dammage the installer script for others.

Maybe someone can update the script
im looking also for llvm-12 version support. script doesnt find llvm-8 but still looking how to solve that one. See other issue

Greetings
Gerald

Eyantra698Sumanto, 2021-07-08:
The whole system of eSim is designed to work using  kicad-4.0.7. Newer versions of kicad not supported.

Eyantra698Sumanto, 2021-07-08:
We would include KiCAD 5.1 in newer release of eSim. Please let us know if you would like to contribute.

Eyantra698Sumanto, 2022-11-01:
@geraldsweers Hi. Any update? Would you like to contribute?

satyap99, 2023-02-22:
@Eyantra698Sumanto I tried installing this pkg on Ubuntu 22.04 and encountered multiple errors
1) Kicad 4.0.7 incompatibility with the install script so I changed it to the latest Kicad-6 and it worked then I faced the 2nd problem 2) Incompatibility with llvm-9 which is not available by default on the latest Ubuntu version. So I tried to change the nghdl.sh script to change the value of variable "llvm_version" to "14" instead of "9". Even after changing it defaults to version 9 of llvm. Can you tell me if there is any other function that's not letting me overwrite the "llvm_version" variable.

Eyantra698Sumanto, 2023-02-22:
@satyap99 Thanks for writing to us. 

Please remove llvm-9 first by running the following command: `sudo apt-get purge -y llvm-9 llvm-9-dev` and then run the eSim/NGHDL install script. Please let us know if this doesn't work.

Also, it is great that you were able to install Kicad-6. We are in the progress of shifting to Kicad-6 in the next version of eSim. Thank You.

satyap99, 2023-02-22:
@Eyantra698Sumanto Actually while running the script it's not able to install llvm-9 as the package in the current version of Ubuntu. Even after changing the llvm_version to 14 in NGHDL script I don't know how it gets overwritten by llvm-9 when I'm running the main install script.

Eyantra698Sumanto, 2023-02-22:
@satyap99 Please see line 61 of the [install-eSim.sh](https://github.com/FOSSEE/eSim/blob/installers/Ubuntu/install-eSim.sh). There is a compressed file named **nghdl.zip** where the install-nghdl.sh exists. When you run install-eSim.sh the nghdl.zip gets extracted and overwrites the install-nghdl.sh which you have changed.

So, please comment on the 61st line of the [install-eSim.sh](https://github.com/FOSSEE/eSim/blob/installers/Ubuntu/install-eSim.sh), make the required changes to the **install-nghdl.sh**, and run the installer again. Please let us know if this doesn't work.