FOSSEE/eSim #166: Support for installing on other Linux distro?

kevinsmia1939, 2021-07-25:
Hello,

I see that this provide installer for Ubuntu, but would it be possible to make an installer for other non apt, non deb distro?
I also would like to create eSIM package as Flatpak.
https://www.flatpak.org/
and distribute it in Flathub.
https://flathub.org/home

Flatpak can install on any distro that support them, which most distro does. KiCAD is already in Flathub. 
https://flathub.org/apps/details/org.kicad.KiCad

I am trying to understand install-esim.sh at the moment.
Would you interest for eSIM as Flatpak?

rahulp13, 2021-07-25:
Hello Kevin,

We would be glad to build eSim through Flatpak (or AppImage) and make it available on all Linux distros. @Eyantra698Sumanto can further help you with the same. Regarding the KiCad version available on Flathub, it is not fully compatible with eSim as we have customized libraries based on KiCad v4.0.7 suitable for eSim.

Eyantra698Sumanto, 2022-10-31:
@kevinsmia1939 @whitelynx Any update?? Would you like to help us by contributing?

kevinsmia1939, 2022-10-31:
Nothing at the moment, but thanks for reminding will look into it.

Eyantra698Sumanto, 2022-11-01:
@kevinsmia1939 Yes please. It would be great!!!

jkrogers9862, 2024-07-11:
What are the obstacles that prevent this from being used on say Fedora or other distros? Is there anything outright that is only available on Ubuntu and Windows? It is using cross platform tech like python which should not be a problem.

Eyantra698Sumanto, 2024-07-11:
Hi @jkrogers9862 , thanks for reaching out. So, we have an installer for the Fedora. Please refer here: https://github.com/FOSSEE/eSim/tree/installers/Fedora. However, it is not maintained much. It will be great if you could help!

jkrogers9862, 2024-07-11:
> Hi @jkrogers9862 , thanks for reaching out. So, we have an installer for the Fedora. Please refer here: https://github.com/FOSSEE/eSim/tree/installers/Fedora. However, it is not maintained much. It will be great if you could help!

Ah I see this now, it is specific to Non-immutable OS variants of Fedora, as such rpm-ostree based immutable distributions won't work. As long as most all dependencies can be installed via dnf then it should be an easy fix. I might take a look if I get a chance.

Eyantra698Sumanto, 2024-07-11:
Yeah, right. We are looking for people who can help us with the Flatpak. There was some progress done: https://github.com/FOSSEE/eSim/pull/244. 

Please let us know if anything.