FOSSEE/eSim #211: Automate the packaging of eSim

Eyantra698Sumanto, 2022-07-24:
### Description

Currently the packaging in eSim is done manually.

However, it is difficult to package everything and follow the steps as new features get added on.

It also takes huge amount of time and the developer also needs to go through a proper checklist.

We need to automate the packaging of eSim to solve the above problems.

### Steps to Reproduce

Automate the packaging of eSim.

### Expected Behavior

A script/code should package the eSim automatically and the developer must have to do minor changes to the script whenever there are change in modules.

### Version and OS

 - eSim-2.2 onwards
 - OS:  Windows (whichever supported by eSim)

saicharan0112, 2022-10-04:
@Eyantra698Sumanto What do you mean by packaging (I mean which branches are packed together)? I have been developing CI for a few opensource tools. Maybe I can help you in packaging and releasing the eSim automatically.

Eyantra698Sumanto, 2022-10-04:
@saicharan0112 Please have a look at the following repository: https://github.com/FOSSEE/eSim/tree/installers/Windows

We are planning to automate the packaging of eSim Windows. Currently, the installers for the same are created manually and then passed to NSIS. But we want to automate the process. 

Please give us your valuable comments!!!

Thank You.

saicharan0112, 2022-10-04:
@Eyantra698Sumanto This looks pretty much straight forward to me. It can be done using GHA without any hustle. Do you want me to try for Ubuntu?

Eyantra698Sumanto, 2022-10-05:
@saicharan0112 Yes it would be great if you could try and let us know!!!

Eyantra698Sumanto, 2022-10-23:
@saicharan0112 If you have any update, please let us know.

saicharan0112, 2022-10-23:
@Eyantra698Sumanto Definitely. I will give an update this week. I was busy with my office work all these days.

saicharan0112, 2022-10-30:
@Eyantra698Sumanto Here is the release that is automatically done by GHA - https://github.com/saicharan0112/eSim/releases/tag/test_tag1

Can you check whether it has all the required files?

If so, I can create a PR so that you people can use it for eSim Ubuntu release

Eyantra698Sumanto, 2022-10-30:
@saicharan0112, Thanks a lot. I quickly went through it. Can you please provide us with a doc or readme mentioning the steps on how you automate it using GHA. Also, please correct the file name "nghld.zip" to "nghdl.zip". We will look into other files carefully and give some more comments in some time.

@rahulp13 feel free to comment.

saicharan0112, 2022-10-30:
@Eyantra698Sumanto Checkout these jobs - https://github.com/saicharan0112/eSim/actions/runs/3356398346/jobs/5561435527

Their names should give you the idea of steps.. This GHA should take care of all the steps you mentioned in this readme file https://github.com/FOSSEE/eSim/tree/installers/Ubuntu

saicharan0112, 2022-10-30:
It also create a release whenever you create a tag starting with "v"

Eyantra698Sumanto, 2022-10-31:
@saicharan0112 Thanks Sai. It looks quite good. Yes, please create a pull request. Also, for Windows, will the GHA be feasible? We are using the NSI script for installation and all stuff.

saicharan0112, 2022-10-31:
@Eyantra698Sumanto I would like to point out two things in this improvement, which I hope you already observed.
* It doesn't check for eSim executable. It directly copies the source code (src/ folder) from nghdl and eSim. 
* I couldn't find any eSim_manual.pdf file in the library/figures something location. However, I observed that we are deleting all the content except this pdf (if present). Is that what you would expect?

Eyantra698Sumanto, 2022-10-31:
@saicharan0112 

- Yay correct. eSim doesn't have an executable binary as of now. This approach should be correct. However, I need to install them to test them.
- Earlier eSim had an HTML-based manual. However, we switched on to LaTeX now. So, the eSim_manual.pdf needs to be there. Here is the link to the eSim manual: https://static.fossee.in/esim/manuals/eSim_Manual_2.3.pdf

rahulp13, 2022-10-31:
@saicharan0112  your gha for Ubuntu looks promising. Some tweaks may be required, but you can proceed with the PR. Regarding the manual, we can automate it by fetching it from the above URL pattern.

saicharan0112, 2022-10-31:
We could use wget to download the manual or build the manual in gha too...
Where the tex files are stored in this repo?

rahulp13, 2022-10-31:
Please check the [manual branch](https://github.com/FOSSEE/eSim/tree/manual). Note that `main.tex` is the starting file.

saicharan0112, 2022-10-31:
Can you provide the list of commands you all usually follow to build the pdf file using these tex files?

Eyantra698Sumanto, 2022-10-31:
@saicharan0112 Generally we do everything in Overleaf. However please have a look here: https://github.com/aslushnikov/latex-online

saicharan0112, 2022-10-31:
@Eyantra698Sumanto @rahulp13 I am not sure about automating the manual build using GHA cuz it looks like it requires hard-coding the name of latex files and figures, which I usually don't encourage. I see additional files in the manual branch. Can we reorganise the files and images, maybe placing them all in a single folder, so that we can run that folder directly using https://github.com/aslushnikov/latex-online .

saicharan0112, 2022-10-31:
If the manual is first updated in the location https://static.fossee.in/esim/manuals/*, I can just pull "https://static.fossee.in/esim/manuals/eSim_Manual_$tag.pdf" based on the tag you use for release (which is in sync with the release here on github). Hows that?

Eyantra698Sumanto, 2022-10-31:
@saicharan0112: https://static.fossee.in/esim/manuals/eSim_Manual_$tag.pdf should be a good solution.

@rahulp13 what's your opinion?

saicharan0112, 2022-11-02:
Any updates on this?

saicharan0112, 2022-11-02:
> @saicharan0112 your gha for Ubuntu looks promising. Some tweaks may be required, but you can proceed with the PR. Regarding the manual, we can automate it by fetching it from the above URL pattern.

@rahulp13 What do mean by tweaks? Can't those be in this PR?

Eyantra698Sumanto, 2022-11-02:
@saicharan0112 please proceed with https://static.fossee.in/esim/manuals/eSim_Manual_$tag.pdf

saicharan0112, 2022-11-06:
@Eyantra698Sumanto Sure. I was actually waiting for @rahulp13 's comments. I missed his thumbs up reactions. Apparently GitHub won't notify you when emojis are used in conversations. 

I will create a PR once I get some time.