FOSSEE/eSim #227: Auto-detecting the previously installed eSim version

Eyantra698Sumanto, 2022-10-23:
### Description
The current eSim installer does not detect the previously installed eSim version.

This causes various path problems and problems with rewriting the files.

### Expected behavior
The installer should be able to detect the previously installed eSim.

Ask the user whether to remove it or not.

If the user permits, the previously installed eSim must be removed completely and then the installation of the new version of eSim must begin.

### Version and OS
 - eSim-2.3
 - OS: Windows 10

saicharan0112, 2022-10-31:
This part can be useful I guess - https://github.com/FOSSEE/eSim/blob/installers/Ubuntu/install-eSim.sh#L374. If these locations have any eSim traces, a message can be printed for the user confirmation and perform uninstallation part of the same script and begin the new version installation. 
@Eyantra698Sumanto What do you say?

Eyantra698Sumanto, 2022-10-31:
@saicharan0112 Thanks for letting me know. However, this issue is basically for the Windows OS...

092vk, 2024-11-14:
I have solved this problem by making changes to the esim-setup-script.nsi script in windows folder in Installers branch . But how can i test my NSIS script , in readme.md of windows two options are listed - [esim-executable](https://github.com/092vk/eSim/tree/installers/Windows#esim-executable) and [How to package eSim](https://github.com/092vk/eSim/tree/installers/Windows#how-to-package-esim) . Which one should i follow to test my esim-setup-script.nsi script. Do i have to repackage to test it .

Eyantra698Sumanto, 2024-11-15:
@092vk At first, the eSim executable must be created using the [esim-executable](https://github.com/092vk/eSim/tree/installers/Windows#esim-executable) script. 

Then, the created executable needs to be added and packaged as given in the [How to package eSim](https://github.com/092vk/eSim/tree/installers/Windows#how-to-package-esim). The NSIS script needs to be run to create the eSim Installer using the packages. Now, the eSim can be installed and tested using the newly created installer.