FOSSEE/eSim #235: Error while installation on Linux Ubuntu

rjrishabh01, 2023-01-16:
local/lib/python3.10/site-packages (2.2.1)
Installing Hdlparse........................
Defaulting to user installation because normal site-packages is not writeable
Collecting hdlparse
  Using cached hdlparse-1.0.4.tar.gz (10 kB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [1 lines of output]
      error in hdlparse setup command: use_2to3 is invalid.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

Error! Kindly resolve above error(s) and try again.

Aborting Installation...

Please guide how to rectify the issue

rahulp13, 2023-01-17:
Which eSim version and Linux OS are you using?

Padmesh-pro, 2023-01-26:
The same error for me I'm using eSim2.3 and Ubuntu 22.04

Eyantra698Sumanto, 2023-01-26:
@Padmesh-pro @rjrishabh01 Please run the following command manually and then rerun the eSim install commands again.
`
pip3 install --upgrade https://github.com/hdl/pyhdlparser/tarball/master`

Eyantra698Sumanto, 2023-02-15:
Solved in the following pull request #231.

closed 2023-02-15 by Eyantra698Sumanto