FOSSEE/eSim #236: eSim crashing upon converting subcircuit to nspice.

hrittikag, 2023-02-07:
### Description

While creating a subcircuit, when we try to convert the kicad model to ngpsice, eSim crashes and becomes unresponsive.

### Steps to Reproduce

### Expected Behavior

Expected eSim to successfully convert kicad model to ngspice and create the subcircuit.

### Screenshots

[image]

### Version and OS

 - eSim version: eSim 2.2
 - OS: Windows 10, 64 bit

rahulp13, 2023-02-08:
@hrittikag, please specify the steps to reproduce the error, right from launching eSim till one gets the error.

Eyantra698Sumanto, 2023-02-15:
Solved in the following pull request #231.

Eyantra698Sumanto, 2023-09-02:
eSim crash when attempting to open or edit a Subcircuit schematic without selecting a project fixed in #256.