FOSSEE/eSim #267: Missing or Unavailable "Switch" component

vishnunithyasoundhar, 2023-09-03:
### Description
Basic switch component is missing or unavailable in the eSim library, in the Schematics designer. 

I tried searching the forums, but it wasn't mentioned anywhere oddly.

[image]
The above is an image of the mentioned switch component.

### Screenshots
[image]

### Version and OS
 - eSim version 2.3
 - OS: Windows 10

### Additional Details
I am submitting this under the guidance of Dr. Maheswari R, Vellore Institute of Technology, Chennai
Name: VishnuNithyaSoundhar S
ID: 23MCS1021

Eyantra698Sumanto, 2023-09-03:
@vishnunithyasoundhar eSim does not support Real-Time Simulation hence a switch is as good as a closed circuit or an open circuit.

In order to implement the functionality of a switch, you may use the aswitch under eSim_Analog libraries. You may input the Pulse or a PWL source at the input terminal of the aswitch to control the switch where the high means switch is closed, and the low means switch is open. Try implementing the same in this way.

If you have some other thing in mind to implement the switch, please feel free to share here.

Please feel free to ask any more queries.

vishnunithyasoundhar, 2023-09-04:
> @vishnunithyasoundhar eSim does not support Real-Time Simulation hence a switch is as good as a closed circuit or an open circuit.
>
> In order to implement the functionality of a switch, you may use [...]

Thank you for your guidance. I will use aswitch in place of switch for the schematic 🙏