FOSSEE/eSim #272: eSim-2.4 unable to convert digital circuit from kicad to NgSpice

ashwini0921, 2024-03-19:
### Description

I tried a simple digital AND gate circuit, I am unable to convert KiCad to NgSpice.
### Steps to Reproduce

[image]
[image]
[and_test-20240319T082955Z-001.zip](https://github.com/FOSSEE/eSim/files/14647675/and_test-20240319T082955Z-001.zip)

1) Unzip the and_test file
2) In eSim open project, locate the and_test folder
3) click convert KiCad to NgSpice option 
### Expected Behavior

A dialog box should appear showing source details, type of analysis
### Screenshots

### Version and OS
ubuntu 20.04.6 LTS

 - eSim version - 2.4
 - OS: [e.g. Windows 10, 64 bit]
Ubuntu 20.04.6
### Additional Context

rudrakshpeps, 2024-05-31:
Same issue i am also facing on windows 11

Eyantra698Sumanto, 2024-09-07:
@rudrakshpeps For now can you try out eSim-2.3 on Windows 11. We are fixing the above issue in the next release.