FOSSEE/eSim #277: Issue while installing eSim 2.3 on Ubuntu

rudrakshpeps, 2024-05-30:
### Description
### Steps to Reproduce

In eSim 2.3 folder 
Opened the terminal 
executed the following command
$ chmod +x install-eSim.sh 
$ ./install-eSim.sh --install

### Expected Behavior
Installation complete without any issue

### Screenshots
[image]

### Version and OS

 - eSim version : eSim-2.3
 - OS: [e.g. Windows 10, 64 bit] Ubuntu Lts 24.04

### Additional Context

Eyantra698Sumanto, 2024-06-14:
@rudrakshpeps eSim-2.4 is not supported on Ubuntu 24.04. It is supported on Ubuntu20.04 and Ubuntu 18.04. Would you like to help us in this regard?

avinashlalotra, 2024-09-06:
@Eyantra698Sumanto I attempted to resolve the installation issue on Ubuntu 24.04. Since python3-distutils is deprecated in Python 3.12 (the version used by Ubuntu 24.04), we can use setuptools as a replacement.

After installing other required packages manually - there’s another issue: esim relies on an older version of KiCad that is no longer available from any host. We will need to manually build this version of KiCad and, most likely, host the package ourselves.

I tried to build the kicad but got few errors.

Eyantra698Sumanto, 2024-09-07:
@avinashlalotra Thanks for your comment. It would be great if you can contribute. Can you create a PR on whatever fixes you have done?

Did you try installing KiCad-6 using the following command:

```
sudo add-apt-repository ppa:kicad/kicad-6.0-releases
sudo apt update -y
sudo apt -y install kicad
```

Please feel free to reply here.

avinashlalotra, 2024-09-07:
@Eyantra698Sumanto You are Welcome .
The method you provided for installing KiCad 6.0 isn't functioning on Ubuntu 24.04, although it works on 20.04 and 22.04. Currently, I’ve encountered three issues with Ubuntu 24.04:
```
1. KiCad Installation: The installation process for KiCad is problematic on this version.
2. System-Wide Package Installation: It's not possible to install system-wide packages using pip in Ubuntu 24.04. 
 Packages should be installed via apt or another appropriate method. Alternatively, this can be managed within a Python virtual environment.
3. LLVM-9 Installation: Installation of LLVM-9 fails during the nghdl sh script. 
```

Eyantra698Sumanto, 2024-09-07:
@avinashlalotra Yes, the problem exists and the dependencies get deprecated for every new version of Ubuntu.

We were trying to get everything on Flatpak. There was some progress done at #244. It will be great if you could help us in completing it.

avinashlalotra, 2024-09-08:
@Eyantra698Sumanto I'll do my best to help with this issue. 
I'm new to Flatpak, but I'll familiarize myself with it and aim to make a meaningful contribution.

avinashlalotra, 2024-11-16:
@Eyantra698Sumanto 
I have successfully managed to get esim running on Ubuntu 23.04. During the process, I encountered several issues and resolved them as follows:

## Issues Encountered & Resolutions:
# LLVM 9 Compatibility:
Ubuntu 23.04 does not have LLVM version 9 available in its repositories. To work around this, I built LLVM 9 locally. While this works for local setups, it's not ideal for production environments.

# Verilator and GHDL Issues:
There were minor compatibility issues with both Verilator and GHDL. These have been addressed and resolved, ensuring that esim now runs successfully on my system.

## Next Steps:
I plan to submit a clean-up PR by the end of next week, which will include a script for automating the build process.
Note: Building LLVM 9 manually is required for the current solution, but this is time-consuming and not suitable for production pipelines.

## Exploring LLVM 18:
In the meantime, I am investigating the possibility of compiling GHDL with LLVM 18 as a more modern and potentially compatible solution. However, I cannot confirm yet whether this will work, as further testing is required.
## Conclusion:
I will update the PR with a script to automate setup once the clean-up is complete.
If anyone is interested in testing with LLVM 18 or has suggestions for optimizing this process, feel free to reach out.

Eyantra698Sumanto, 2024-11-16:
Thanks a lot @avinashlalotra for the updates!!! It's a great progress.

@jayanth-4547 is also looking into the issue. Maybe you both can collaborate.

If required, it is better to explore higher versions of GHDL and LLVM. Building from source is not a feasible solution as it would take a lot of time and consume a lot of resource, build issues.

If anything, please feel free to reply in this thread.

Thanks!!!

Jayanth-4547, 2024-11-18:
@avinashlalotra 
Hi, this is Jayanth.

I’m currently working on the same issue and have managed to successfully run eSim as well. 
For my setup, I have used **llvm_version="15"** and **ghdl="ghdl-gha-ubuntu-22.04-llvm"**, and everything seems to be working fine. 

Currently, I’m addressing an issue with the centering of the dialog boxes and the eSim application itself, as they are positioned at the top-right corner.  
Here’s an example:  
[image]  

Have you experienced the same issue? If so, I’d appreciate any insights you might have on resolving it.  

Additionally, I’m curious to know how you’ve handled the **Hdlparse**, **Makerchip**, and **SandPiper SaaS** modules.  

Looking forward to your response!

avinashlalotra, 2024-11-21:
@Jayanth-4547  
Hi Jayanth,

Regarding the issue with centering the dialog boxes, I am not familiar with the solution. I suggest reviewing the codebase, as I believe you will be able to identify the approach.

You can install hdlparse and other required packages within a virtual environment. It is also recommended to run eSim in that environment for consistency.

I am currently working on a  installation script of eSim . Focusing our efforts on Ubuntu 23 should allow eSim to be released for Ubuntu 23 by the end of this year.

avinashlalotra, 2024-12-01:
@Eyantra698Sumanto 
Hi Sumanto,

Could you please test the following repository for installing eSim 2.4 on Ubuntu 23?
https://github.com/avinashlalotra/esim_ubuntu_installation

This should work without any issue with llvm version. 
And I used GHDL 4.1.0 in it along with minor changes in verilator source . It will succesfully build with the modified script.

I've already tested it on Ubuntu 22 and 23. If you encounter any issues, kindly share your feedback so I can address them and push the necessary fixes.

Thank you!

Eyantra698Sumanto, 2024-12-02:
Thanks a lot Avinash! We will check this in sometime and get back to you. Thanks again.

Eyantra698Sumanto, 2025-01-11:
Thanks @AbhijitBaral . Will update this.

Eyantra698Sumanto, 2025-01-11:
@AbhijitBaral Also you may contribute to the repository by creating a PR. Please feel free to create a PR.

AbhijitBaral, 2025-01-11:
I checked. The necessary changes are already present..

Aishw31, 2025-03-14:
Hi @Eyantra698Sumanto 

> [@Eyantra698Sumanto](https://github.com/Eyantra698Sumanto) Hi Sumanto,
>
> Could you please test the following repository for installing eSim 2.4 on Ubuntu 23? [...]

Hi @Eyantra698Sumanto,

@avinashlalotra and I collaborated on this repository to fix the issues we faced while installing eSim on Ubuntu 23. The changes have been successfully merged into the installer_branch.

It all started when I applied for the FOSSEE Winter Internship 2024. Unfortunately, I wasn’t selected, but that didn’t demotivate me. Instead, @avinashlalotra and I successfully fixed the issues.

This year, I am applying for the FOSSEE Summer Fellowship, and I hope to make it this time. While going through the screening tasks, I noticed that the second task is the same issue we had already fixed. However, there might be other issues we didn’t encounter. As far as I know, it was working on Ubuntu 23/22.

Should I submit this task again, or can my previous contribution be considered?

avinashlalotra, 2025-03-14:
Hi @Aishw31,

It was fun collaborating with you! I hope you make it into the summer fellowship this time. Wishing you all the best!

FOSSEE is a great organization to work with. I was a Semester-Long Intern in 2024 and am still working on publishing the Signal Processing Toolbox. The mentors are very supportive.

Eyantra698Sumanto, 2025-04-08:
Hi @Aishw31, sorry for the delayed reply. Thanks for commenting here. We will consider your previous submission as the Fellowship task. Please send an email to contact-esim@fossee.in. Best of luck!

Aishw31, 2025-04-09:
Hi @Eyantra698Sumanto, Thank you for your response.
I have mailed it to the specified address. I'm looking forward to contributing more to eSim.

PR #341 (2025-05-25): Installer script updated to detect and support Ubuntu 22.04, 23.04, and 24.04 during installation
### Related Issues

Fixes #277

### Purpose

The installer script lacked the capability to detect and handle different Ubuntu versions (22.04, 23.04, and 24.04), which could cause installation failures or improper behavior on these systems. This update adds version detection to improve compatibility and user experience.

### Approach

- Enhanced the installer script to detect Ubuntu versions 22.04, 23.04, and 24.04.
- Added conditional handling to support installation steps specific to these versi [... cut]
commit 5088e227 (2025-05-25): Fixes #277 - Add Ubuntu version detection to installer script to handle versions 22.04, 23.04, and 24.04