FOSSEE/eSim #282: error: externally-managed-environment

Shiva0krishna, 2024-11-04:
### Description
Installations in the python environement in  Ubuntu  is conflicting  system-wide installations.

### Steps to reproduce
Navigate to eSim directory :
           cd eSim-2.4
Make the Installation Script Executable:
          chmod +x install-eSim.sh
Run the Installation Script:
         ./install-eSim.sh --install

Comment out the code

           echo "Installing Distutils......................."
          sudo apt-get install -y python3-distutils  

cause this is yet an other error needs to solved.

### Expected behavior
Installation of watchdog 
### Screenshots
[image]

### Additional files
[image]

### Version and OS
 - eSim version  -  eSim-2.4
 - OS: [e.g. Windows 11, 64 bit, Ubuntu 24.04 LTS]

092vk, 2024-11-13:
Why this issue is happening :
Recent versions of Ubuntu (22.04 and later) and Debian have adopted a policy to prevent direct use of pip to install Python packages system-wide. So as we can see in error , it is suggesting that if you want to install a non-debian packaged python package you have to do it using apt , use of pip is discouraged in global context as mentioned above to avoid conflict with apt . 

Solution : 
As mentioned in log itself you can use tools like pipx or create a virtual environment to avoid any conflict.

Using pipx : pipx install Watchdog(any other package)

Using venv (Recommended):

1.Install venv package if not already present in your Ubantu 
--> sudo apt install python3-venv

Navigate to eSim directory
-->cd eSim-2.4

2.Create a Virtual Environment named eSim
--> python3 -m venv eSim

3.Activate the Virtual Environment 
--> source eSim/bin/activate

4.Make the Installation Script Executable:
-->chmod +x install-eSim.sh

5.Run the Installation Script:
-->./install-eSim.sh --install

Eyantra698Sumanto, 2024-11-13:
> @Eyantra698Sumanto <https://github.com/Eyantra698Sumanto>
> @Shiva0krishna <https://github.com/Shiva0krishna>

Thanks Vivek for commenting!

Thanks & Regards,
Sumanto Kar