FOSSEE/eSim #284: Unhandled version llvm 11.1.0

Shiva0krishna, 2024-11-17:
### Description
**Error  encountering indicates that the script is unable to find the llvm-9 and llvm-9-dev packages. This is because, by default, Ubuntu 22.04 may not have LLVM 9 available in its package repositories**

```
Installing LLVM-9........................................
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package llvm-9
E: Unable to locate package llvm-9-dev

Error! Kindly resolve above error(s) and try again.

Aborting Installation... 
```

> Considerable changes :  llvm_version="9" to llvm_version="11"

**_Outcome : llvm-11 packages has installed succesfully however,  GHDL is not compatable  with the version llvm 11.1.0_**

```
Installing ghdl-0.37 LLVM.................................
ghdl-0.37 successfully extracted
Changing directory to ghdl-0.37 installation
Configuring ghdl-0.37 build as per requirements
Use full IEEE library
Build machine is: x86_64-linux-gnu
Unhandled version llvm 11.1.0

Error! Kindly resolve above error(s) and try again.

Aborting Installation...
```

### Steps to reproduce
Navigate to eSim directory :
cd eSim-2.4
Make the Installation Script Executable:
chmod +x install-eSim.sh
Run the Installation Script:
./install-eSim.sh --install
change the llvm version to 11  in  nghdl.sh folder 

### Expected behavior
Installation of llvm which is compatable with GHDL

### Screenshots
[image]

### Version and OS
 - eSim version -2.4
 - OS:Ubuntu 22.04

### Additional context
I think   considering GHDL upgradtion  is neccessary.

avinashlalotra, 2024-12-01:
@Shiva0krishna 
I upgraded it with  GHDL  4.1.0 . 
Apart from that there were some headers file missing for verilator. 
Please try installing eSim 2.4 on Ubuntu 22/23 using this repo https://github.com/avinashlalotra/esim_ubuntu_installation
And provide me feedback So I will push this to respective branches . I tested this and it is working on my system . 

Thank You

Eyantra698Sumanto, 2024-12-29:
@avinashlalotra please create a PR for your installer on the eSim installer branch: https://github.com/FOSSEE/eSim/tree/installers

avinashlalotra, 2024-12-31:
@Eyantra698SumantoThe changes that fix  this issue #284 are in the file install-nghdl.sh, which is inside the nghdl.zip folder. I tried looking for the repository that contains the install-nghdl.sh file, but I couldn't find it. It would be helpful if you could guide me on where to make these changes

Eyantra698Sumanto, 2024-12-31:
@avinashlalotra The nghdl branch is maintained separately. It is available here: https://github.com/FOSSEE/nghdl/blob/installers/Ubuntu/install-nghdl.sh. Please let me know if anything.

avinashlalotra, 2025-01-01:
@Eyantra698Sumanto  

I created two PRs in the esim-installers and nghdl-installers repositories. Please review them.
Thank you.

Eyantra698Sumanto, 2025-01-01:
@avinashlalotra Thanks! Will review and merge them.