FOSSEE/eSim #294: kiCad to ngspice window doesnot open

AbhijitBaral, 2025-01-11:
### Description
After exporting netlist from eeschema, the kiCad to ngspice converter window doesnot open when the appropriate option is clicked

### terminal flow trace
```
Function: Validating New Project Information
Function : Open Kicad Schematic
Function : Validating for Kicad components
Worker Thread Calling Command : eeschema /home/abhijit/eSim-Workspace/test/test.kicad_sch
Function: Open Kicad to Ngspice Converter
Function : Validating for Kicad components
==================================
Kicad to Ngspice netlist converter
==================================
.title KiCad schematic
vin1 input GND sine
ip_voltage1 input plot_v1
R2 GND output 10k
R1 input output 10k
op_voltage1 output plot_v1
.end
Traceback (most recent call last):
  File "/home/abhijit/eSim-2.4/src/projManagement/Kicad.py", line 213, in openKicadToNgspice
    self.obj_dockarea.kicadToNgspiceEditor(var)
  File "/home/abhijit/eSim-2.4/src/frontEnd/DockArea.py", line 228, in kicadToNgspiceEditor
    self.kicadToNgspiceLayout.addWidget(MainWindow(clarg1, clarg2))
  File "/home/abhijit/eSim-2.4/src/kicadtoNgspice/KicadtoNgspice.py", line 95, in __init__
    schematicInfo, sourcelist = obj_proc.insertSpecialSourceParam(
  File "/home/abhijit/eSim-2.4/src/kicadtoNgspice/Processing.py", line 148, in insertSpecialSourceParam
    if words[3] == "pulse":
IndexError: list index out of range
```
### Version and OS  
 - eSim version : 2.4
 - OS: Ubuntu 20.04.6
-  KiCad: 6.0.11-2627ca5db0~126~ubuntu20.04.1, release build

092vk, 2025-01-11:
Is it happening with a particular circuit or all? If particular can you mention it and share its image .

AbhijitBaral, 2025-01-12:
What I figured out is, it happens when the "reference" of an e-sim plot element is changed from it's default "Ux" value. 
Also I've had problems during simulation, when I changed the reference name of MOSFETS in a schematic..

AbhijitBaral, 2025-01-12:
So it is best to leave the reference names of elements in KiCad schematics as they are by default.

closed 2025-01-13 by AbhijitBaral